STMicroelectronics/STM32CubeF7 #80: STM32F769 - STOP Mode + LCD / DSI - Screen not working on wake up

ladislas, 2022-12-06:
Hi there! 

Our product using and STM32F769 + OTM8009A LCD screen (the same as the DISCO_F769) has been working great for quite some time now. We've been using all the drivers provided by ST to have the LCD run and display images or play videos.

Recently, we have been working on enabling `deep sleep` in our product to reduce power consumption when it's not used.

We've discovered that when exiting `deep sleep` the screen start behaving strangely until it completely stops working.

It looks like this:

[image]

Colors are shifting when we turn off (deep sleep) and turn on (wake up) the video. Position is also shifting. If you do that in a loop, sometimes it "works again" then fails again until it eventually crashes completely.

The PR adding the bug is here:

https://github.com/leka/LekaOS/pull/1173/files

Our guess is that there are issues with the DSI or LTDC clock when deep sleeping / waking up, but we don't even know where to start looking...

Any ideas? :slight_smile: 

Thanks for the help!

Best,
-- Ladislas

jeromecoutant, 2022-12-06:
Note that `deep sleep` in this case is STOP mode.

HBOSTM, 2023-02-06:
Hello @ladislas ,

Could you please confirm that you are using the drivers provided by ST from [here](https://github.com/STMicroelectronics/STM32CubeF7/tree/master/Drivers/STM32F7xx_HAL_Driver). If so, give us more details about how to reproduce this issue?

Best Regards,

HBOSTM, 2023-05-04:
Hello @ladislas 

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2023-05-04 by HBOSTM