STMicroelectronics/STM32CubeF7 #85: Settings of the main internal Regulator output voltage is not working

KamilDuljas, 2023-03-20:
Nucleo-F756ZG (MB1137 Rev B)
MXCubeIde 1.1.12 GNU Tools for STM32 10.3-2021.10

After call LL_PWR_SetRegulVoltageScaling(LL_PWR_REGU_VOLTAGE_SCALE3) wait for PWR_CSR1_VOSRDY bit.
When call LL_PWR_IsActiveFlag_VOS() then wait in infinite loop because PWR_CSR1_VOSRDY bit is always 0
It is true for all LL_PWR_REGU_VOLTAGE_SCALEx values.

**How To Reproduce**
1. Create default project
2. In ioc file, Project manager-> advanced settings -> driver sellector for "RCC" and "Cortex M7" choose LL instead of HAL
3. Generated function SystemClock_Config() looks like this:

```
void SystemClock_Config(void)
{
  LL_FLASH_SetLatency(LL_FLASH_LATENCY_0);
  while(LL_FLASH_GetLatency()!= LL_FLASH_LATENCY_0)
  {
  }
  LL_PWR_SetRegulVoltageScaling(LL_PWR_REGU_VOLTAGE_SCALE3);
  while (LL_PWR_IsActiveFlag_VOS() == 0)
  {
  }
  LL_PWR_DisableOverDriveMode();
  LL_RCC_HSI_SetCalibTrimming(16);
  LL_RCC_HSI_Enable();

   /* Wait till HSI is ready */
  while(LL_RCC_HSI_IsReady() != 1)
  {

  }
  LL_RCC_SetAHBPrescaler(LL_RCC_SYSCLK_DIV_1);
  LL_RCC_SetAPB1Prescaler(LL_RCC_APB1_DIV_4);
  LL_RCC_SetAPB2Prescaler(LL_RCC_APB2_DIV_2);
  LL_RCC_SetSysClkSource(LL_RCC_SYS_CLKSOURCE_HSI);

   /* Wait till System clock is ready */
  while(LL_RCC_GetSysClkSource() != LL_RCC_SYS_CLKSOURCE_STATUS_HSI)
  {

  }
  LL_SetSystemCoreClock(16000000);

   /* Update the time base */
  if (HAL_InitTick (TICK_INT_PRIORITY) != HAL_OK)
  {
    Error_Handler();
  }
}
```

HBOSTM, 2023-03-22:
Hi @KamilDuljas , 

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension. 

With regards,

HBOSTM, 2023-03-22:
ST Internal Reference: 148076

KamilDuljas, 2023-03-22:
Thanks. Othermore, the question is:
If according to documentation stm32f75xxx the reset value of pwr_cr1 register is 0x0000 C000 so means that VOS[1:0] is set to 1 1, Then pwr_csr1->vosrdy should been set to 1 far long ago after initialization, for example in while loop, without use above macros?

It looks like a hardware problem related to F7 or nucleo board

TOUNSTM, 2023-04-10:
Hello @KamilDuljas,

Thank you for your contribution. You are absolutely right about this point. Actually, the point you raised has already been dealt and fixed internally. The fix will be made available in the frame of a future release of CubeMX.

Unfortunately we don't treat aspect related to CubeMX tool in our GitHub repositories. They are rather treated at the [STM32CubeMX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated page of the ST Community.

Since this issue is not directly related to the STM32Cube firmware but rather to our ecosystem, please allow me then to close this thread. Thank you for your comprehension.

With regards,

closed 2023-04-10 by TOUNSTM