STMicroelectronics/STM32CubeF7 #87: HAL_DMA_Init set invalid state when DMA_CheckFifoParam not return HAL_OK

KamilDuljas, 2023-04-01:
https://github.com/STMicroelectronics/STM32CubeF7/blob/6a3b9bb8d09839f365ba83a6cf5823ae5f5226db/Drivers/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal_dma.c#L279

In related to documentation, The checking compatibility before FIFO threshold level and size of the memory burst is very fragile but after returned other state than HAL_OK the function set HAL_DMA_STATE_READY instead of ERROR or RESET.

MCU H7 is also affected:
https://github.com/STMicroelectronics/STM32CubeH7/blob/b75d4ea0a368c0e2e7bb9c40c1258a39fff0919a/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_dma.c#L341

HBOSTM, 2023-04-04:
Hi @KamilDuljas ,

Thank you for this contribution, this point has been reported to our development teams. I will get back to you as soon as I have any updates.

With regards,

HBOSTM, 2023-04-04:
ST Internal Reference:  149047

TOUNSTM, 2024-06-06:
Fixed in 7f25a108012707cc7a643a6bee79947bd068e0ab