STMicroelectronics/STM32CubeF7 #88: Some text can be more comfortable.

kuailezhizu, 2023-04-06:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board STM32F746G-DISCO
 * I am using IAR 

**Additional context**
I find a project in "STM32Cube_FW_F7_V1.17.0\Projects\STM32746G-Discovery\Examples\LTDC\LTDC_Display_2Layers".
In this workspace and this project, the group Drivers->BSP->STM32F7x6G-EVAL, I think its name should be "STM32746G-Discovery" like other projects. Although it works well, change this text can make some programmer happier(include me).Thank you for reading my issue. 
**Screenshots**
[image]

TOUNSTM, 2023-04-11:
Hi @kuailezhizu,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,

HBOSTM, 2023-04-26:
Hi @kuailezhizu ,

Thank you again for this contribution, this point has been reported to our development teams. I will get back to you as soon as I have any updates.

Best Regards,

ALABSTM, 2024-03-06:
Fixed in commit 7b869e0519bb392f39892db9789d08a27161dfcb

ALABSTM, 2024-03-06:
Hi @kuailezhizu,

I hope you are fine. Just to let you know this issue has been fixed and the fix just been uploaded. Thank you again for having reported.

With regards,

closed 2024-03-06 by ALABSTM