STMicroelectronics/STM32CubeF7 #89: F756: Conitnious mode does not work on ADC with particular settings

KamilDuljas, 2023-04-13:
Nucleo-F756ZG
STM32CubeIDE 1.12.0

I choosed Vrefint channel as model.

When I use ``conversion mode`` - ``EOC flag at the end of single channel conversion`` for ADC1 for one channel, a program read value once and then does not update anymore.

[image]

On screen I highlighted one option: End of conversion selection. When i set  ``EOC flag at the end of single channel conversion`` it doesn't work. When I change it on ``EOC flag at the end of all conversions`` it works ok.

My code:
```
int main(void)
{
  
  HAL_Init();
  SystemClock_Config();
  MX_GPIO_Init();
  
  MX_ADC1_Init();
  MX_USART3_UART_Init();
  
  HAL_ADC_Start(&hadc1);
  
  while (1)
  {
 
      uint32_t value2 = HAL_ADC_GetValue(&hadc1);
      float voltage2 = 3.3f * value2 / 4096.0f;
      printf("Vref = %lu (%.3f V)\n", value2, voltage2);
      
      //__HAL_ADC_CLEAR_FLAG(&hadc1,ADC_FLAG_EOC); // Optional but does not work
      
     HAL_Delay(250);
   }
}
```

In my opinion: ``End Of Conversion Selection`` should not be have impact on continious mode in this case., but even if, it should be work on both options. In manual, there's no information about impact EOC flag on ``continious mode``. We read about ``continious mode``:
[image]

I tested it also on 2 channels on ADC1 - IN0 and Vrefint

KamilDuljas, 2023-04-19:
Any information about this?

RJMSTM, 2023-05-11:
Hello @KamilDuljas,

Sorry for my tardy reply, I'm still investigating I will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension. 

With regards,
Rania

RJMSTM, 2023-05-11:
Hello @KamilDuljas,

Would you please give us more details about how you got this issue? And please share snippet of the code you have used to reproduce this problem.

With regards,

KamilDuljas, 2023-06-01:
Hello I'm recently out of office and I think that I can give update within 30 days.

KamilDuljas, 2023-08-29:
I found the solution:

[image]

Se I must disable overrun feature if i try use continous conversion without dma

closed 2023-08-29 by KamilDuljas