STMicroelectronics/STM32CubeF7 #91: Nucleo-F756zg: Debug support for StopMode does not work

KamilDuljas, 2023-04-19:
Hi, I try to run [PWR_CurrentConsumption example](https://github.com/STMicroelectronics/STM32CubeF7/tree/master/Projects/STM32F746ZG-Nucleo/Examples/PWR/PWR_CurrentConsumption) on Nucleo-756zg.
All works fine without debug, but i tried use each mode in debug mode.
Sleep mode and StandBy mode work fine with debug but StopMode not.

When we try to enter to stop mode and will be executed _WFI() then immidately exit from the mode during debug session. 
I tried perform this through:
STM32CubeIde 1.12 and 
STM32CubeProgrammer

The only one modification against to original exmaple is keep PA13 PA14 PA15 as there are pins connected to stlink.
https://github.com/STMicroelectronics/STM32CubeF7/blob/6a3b9bb8d09839f365ba83a6cf5823ae5f5226db/Projects/STM32F746ZG-Nucleo/Examples/PWR/PWR_CurrentConsumption/Src/stm32f7xx_lp_modes.c#L91
```
GPIO_InitStruct.Pin = GPIO_PIN_All & ~(GPIO_PIN_13 | GPIO_PIN_14 | GPIO_PIN_15);

All usecases executed using multimeter.

RJMSTM, 2023-04-20:
Hello @KamilDuljas ,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

 With regards,
Rania

HBOSTM, 2023-04-26:
Hello @KamilDuljas,

Firstly, thank you for this contribution. This example does not work in debug mode as indicated in the readme file.
https://github.com/STMicroelectronics/STM32CubeF7/blob/6a3b9bb8d09839f365ba83a6cf5823ae5f5226db/Projects/STM32F746ZG-Nucleo/Examples/PWR/PWR_CurrentConsumption/readme.txt#L47
So, please allow me to close this issue and thank you again for your contribution.

Best Regards,

KamilDuljas, 2023-04-26:
Thanks for information.

closed 2023-04-26 by KamilDuljas