STMicroelectronics/STM32CubeF7 #95: SD_read in sd_diskio.c fails with DMA in cached memory areas

Jason0926Ni, 2023-06-02:
When Buffer is not 32-byte aligned, SD_Read will invalidate the Data Cache according to the following code. 
When calculating alignedAddr, the program will add some extra memory addresses to the front, which will cause problems with the data in those memory addresses.

\Middlewares\Third_Party\FatFs\src\drivers\sd_diskio_dma_rtos_template_bspv1.c
\Middlewares\Third_Party\FatFs\src\drivers\sd_diskio_dma_rtos_template_bspv2.c
```

#if (ENABLE_SD_DMA_CACHE_MAINTENANCE == 1)
                /*
                the SCB_InvalidateDCache_by_Addr() requires a 32-Byte aligned address,
                adjust the address and the D-Cache size to invalidate accordingly.
                */
                alignedAddr = (uint32_t)buff & ~0x1F;
                SCB_InvalidateDCache_by_Addr((uint32_t*)alignedAddr, count*BLOCKSIZE + ((uint32_t)buff - alignedAddr));
#endif

```

TOUNSTM, 2023-06-02:
Hello @Jason0926Ni ,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,

ASELSTM, 2023-07-12:
ST Internal Reference: 157163

TOUNSTM, 2024-06-06:
Hello @Jason0926Ni,

First we would like to thank you for your report. In fact, `alignedAddr ` is used only for cache invalidation, it is not used for buffer data read/write.

As this issue is invalid, please allow me thus to close this thread. Thank you fir your comprehension.

With regards,

closed 2024-06-06 by TOUNSTM