STMicroelectronics/STM32CubeF7 #97: Wrong description about HAL_IWDG_DEFAULT_TIMEOUT

almightychang, 2023-07-19:
The comment about `HAL_IWDG_DEFAULT_TIMEOUT` seems to be outdated:
https://github.com/STMicroelectronics/STM32CubeF7/blob/fe78bad77db8206ff7b65e6d0d07fca912756404/Drivers/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal_iwdg.c#L120-L129

The comments suggests to multiply by 1000 to convert into milliseconds. However, it is already in milliseconds, and it is compared with `HAL_GetTick()`.

https://github.com/STMicroelectronics/STM32CubeF7/blob/fe78bad77db8206ff7b65e6d0d07fca912756404/Drivers/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal_iwdg.c#L203

ALABSTM, 2023-08-15:
Hi @almightychang,

Looks like there is a slight misunderstanding. The descriptive comment says that the timeout value **has already been multiplied**  by `1000` to be in milliseconds. **No need** to multiply it again at the user application level.

https://github.com/STMicroelectronics/STM32CubeF7/blob/fe78bad77db8206ff7b65e6d0d07fca912756404/Drivers/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal_iwdg.c#L126

I hope this removes the ambiguity. Please let me know whether this answers your question.

With regards,

ALABSTM, 2023-08-22:
Hi @almightychang,

I hope my previous comment helped clarify the point. Please allow me to close this thread. Thank you again for your post.

With regards,

closed 2024-02-19 by ALABSTM