STMicroelectronics/STM32CubeF7 #98: Wrong information about boot address through option bytes

KamilDuljas, 2023-07-19:
In RM0410, for STM32F76xxx and and STM32F77
xxx boundary of SRAM2  is from 0x2007 C000 to 0x2007 FFFF. In description of Flash register, OPTCR1 in details we read:
BOOT_ADD1[15:0]: Boot base address when Boot pin =1 BOOT_ADD1[15:0] correspond to address [29:14], The boot memory address can be programmed to any address in the range 0x0000 0000 to 0x2004 FFFF with a granularity of 16KB.
Example:
BOOT_ADD1 = 0x0000: Boot from ITCM RAM (0x0000 0000) 
BOOT_ADD1 = 0x0040: Boot from System memory bootloader (0x0010 0000) 
BOOT_ADD1 = 0x0080: Boot from Flash on ITCM interface (0x0020 0000) 
BOOT_ADD1 = 0x2000: Boot from Flash on AXIM interface (0x0800 0000) 
BOOT_ADD1 = 0x8000: Boot from DTCM RAM (0x2000 0000) 
BOOT_ADD1 = 0x8004: Boot from SRAM1 (0x2002 0000) 
BOOT_ADD1 = 0x8013: Boot from SRAM2 (0x2004 C000)

This description is wrong because address 0x2004 FFFF is boundary for SRAM2 on stm32f75 and 74 series. This description is copy/paste from RM0385

RJMSTM, 2023-07-20:
Hello @KamilDuljas,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,

TOUNSTM, 2023-08-02:
ST Internal Reference: 158601

closed 2023-09-11 by TOUNSTM