STMicroelectronics/STM32CubeF7 #104: FreeRTOS: No Memory Pool API Support

MucaGinger, 2023-09-14:
CMSIS RTOS v2 files do not provide support for "Memory Pool" functions. Their declaration is provided inside `cmsis_os2.h` but no definition is provided inside `cmsis_os2.c` source file. Additionally, files `freertos_mpool.h` and `freertos_os2.h` are missing from path `Middlewares\Third_Party\FreeRTOS\Source\CMSIS_RTOS_V2` when compared to STM32CubeF4 firmware.

I suppose there is a reason Memory Pool API isn't provided for STM32F7 MCUs? Is support for Memory Pool API planned anytime in the future?

ALABSTM, 2025-04-25:
See also [STM32CubeF4#170](https://github.com/STMicroelectronics/STM32CubeF4/issues/170)

ALABSTM, 2025-04-25:
Hi @lgacnik,

Please excuse this delayed reply. I let you have a look to [this](https://github.com/STMicroelectronics/STM32CubeF4/issues/170#issuecomment-2129973245) comment that answers your question.

With regards,

closed 2025-04-25 by ALABSTM