STMicroelectronics/STM32CubeF7 #106: Timers: HAL_TIM_PWM_Stop() disable timer even channel 5 or channel 6 are enabled

KamilDuljas, 2023-10-23:
TIM_CCER_CCxE_MASK makro should be contain 6 channels, not  4. Missing `TIM_CCER_CC5E` and `TIM_CCER_CC6E`
[image]

Example:
```
HAL_TIM_PWM_Start(&htim1,TIM_CHANNEL_1);
HAL_TIM_PWM_Start(&htim1,TIM_CHANNEL_2);
HAL_TIM_PWM_Start(&htim1,TIM_CHANNEL_5);

HAL_TIM_PWM_Stop(&htim1,TIM_CHANNEL_1);
// TIM_CR1_CEN is '1' (TIM_BDTR_MOE is '1' for  advanced timers)
HAL_TIM_PWM_Stop(&htim1,TIM_CHANNEL_2);
// TIM_CR1_CEN is '0' but CHANNEL 5 is still active!!!
HAL_TIM_PWM_Stop(&htim1,TIM_CHANNEL_5);
```

The issue related to macros TIM_CCER_CCxE_MASK  and:

```
/**
  * @brief  Disable the TIM peripheral.
  * @param  __HANDLE__ TIM handle
  * @retval None
  */
#define __HAL_TIM_DISABLE(__HANDLE__) \
  do { \
    if (((__HANDLE__)->Instance->CCER & TIM_CCER_CCxE_MASK) == 0UL) \
    { \
      if(((__HANDLE__)->Instance->CCER & TIM_CCER_CCxNE_MASK) == 0UL) \
      { \
        (__HANDLE__)->Instance->CR1 &= ~(TIM_CR1_CEN); \
      } \
    } \
  } while(0)

/**
  * @brief  Disable the TIM main Output.
  * @param  __HANDLE__ TIM handle
  * @retval None
  * @note The Main Output Enable of a timer instance is disabled only if all the CCx and CCxN channels have been
  *       disabled
  */
#define __HAL_TIM_MOE_DISABLE(__HANDLE__) \
  do { \
    if (((__HANDLE__)->Instance->CCER & TIM_CCER_CCxE_MASK) == 0UL) \
    { \
      if(((__HANDLE__)->Instance->CCER & TIM_CCER_CCxNE_MASK) == 0UL) \
      { \
        (__HANDLE__)->Instance->BDTR &= ~(TIM_BDTR_MOE); \
      } \
    } \
  } while(0)
```

Also please look closer `TIM_CCER_CCxNE_MASK`

KamilDuljas, 2023-10-23:
H7 repo is also affected.

KRASTM, 2024-10-28:
Hello @KamilDuljas,

Thank you for the report. 

Regarding the issue, and based on the RM either for F7 or H7:

- For the **TIM_CCER_CCxNE_MASK**: there are no CC4NE, CC5NE and CC6NE, so I think it's ok for this macro.

- For the  **TIM_CCER_CCxE_MASK:** I reported this one to our team in order to analyze it.

With regards,

KRASTM, 2024-10-28:
ST Internal Reference: 195184

TOUNSTM, 2024-11-22:
Fixed in ea67690c22fcf4683e8a1cda948b43cd0c77aa11

closed 2024-11-22 by TOUNSTM