STMicroelectronics/STM32CubeF7 #110: TIM: dirty state SR register after invoke TIM_Base_SetConfig()

KamilDuljas, 2023-12-21:
After invoke TIM_Base_SetConfig SR register contains set update event flag. This flag is generate by line:

https://github.com/STMicroelectronics/stm32f7xx_hal_driver/blob/8c7077398fb8c4481f7d843e25b4f1471971b442/Src/stm32f7xx_hal_tim.c#L6967

It should be cleared by HAL before giving up control to main code.

Proposal fix:
```  
 /* Generate an update event to reload the Prescaler
     and the repetition counter (only for advanced timer) value immediately */
  TIMx->EGR = TIM_EGR_UG;
// Clear flag
TIMx->SR = ~(TIM_IT_UPDATE);
```

ALABSTM, 2023-12-25:
Hi @KamilDuljas,

You are right about the fact that the `UIF` bit in the `SR` is set upon setting bit `UG` in the `EGR` by software. This point will be forwarded to our development teams. They will conclude whether it is a bug that should be fixed or whether this is to serve a certain purpose. I will keep you informed. Thank you for having reported.

With regards,

ALABSTM, 2023-12-25:
ST Internal Reference: 169543