STMicroelectronics/STM32CubeF7 #114: Add example code on PTP

Shuilinguo, 2024-01-10:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
 * IDE or at least the compiler and its version.

**Describe the bug**
A clear and concise description of what the bug is.

**How To Reproduce**
1. Indicate the global behavior of your application project.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).

3. The use case that generates the problem.

4. How we can reproduce the problem.

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

Shuilinguo, 2024-01-10:
Why is there no example code added about time synchronisation?

ALABSTM, 2024-01-12:
Hi @Shuilinguo,

May I suggest that, if you have found such an example within another STM32Cube firmware, you try to port it to the STM32CubeF7? In the meanwhile, your question will be forwarded to our development teams. I will keep you informed.

With regards,

Shuilinguo, 2024-01-15:
Hi,
I didn't find a PTP example in STM32Cube, so I hope you can provide it to STM32CubeF7!

ALABSTM, 2024-01-17:
Hi @Shuilinguo,

To be sure I correctly understood your request, by PTP you mean **P**recision **T**ime **P**rotocol. Don't you?

With regards,

ALABSTM, 2024-01-17:
Hi @Shuilinguo,

According to our development teams, an example using the PTP (**P**recision **T**ime **P**rotocol) shall be made available. Unfortunately, we cannot share a date for the moment.

Regarding the STM32CubeF7 firmware, there is currently no plan to add such an example. Please allow me to close this issue. Thank you for your comprehension and thank you again for having tackled the subject.

With regards,

closed 2024-01-17 by ALABSTM