STMicroelectronics/STM32CubeF7 #115: ETH: Missing ETH_MACCR_CSTF in cmsis header and driver

HTRamsey, 2024-01-20:
CSTF is mentioned in the documentation (taken from RM0410) but not provided in the cmsis header or eth driver. It is, however, correctly added to H7 as ETH_MACCR_CST.
[image]

HTRamsey, 2024-01-20:
Also a quick question, is `ETH_MACDMAConfig` supposed to configure to reset values or are these configurations just arbitrarily chosen? For example, `RetryTransmission` is disabled making RD high instead of its reset value of low. The H7 driver enables it by default, which leaves RD at its reset value of low.

TOUNSTM, 2024-01-23:
ST Internal Reference: 143308

closed 2024-01-28 by HTRamsey

TOUNSTM, 2024-01-29:
Fixed in 30b30b27616d01af3fa11e0ba1662500bb6b5b3a

TOUNSTM, 2024-01-29:
Fixed in 0495bf0fda6f592b034a4267f8e7c7b816e4196e