STMicroelectronics/STM32CubeF7 #116: STM32 USB OTG Dual Role

Shikamru, 2024-02-22:
HI team,
          I am using STM32F7 series controller. I want to use USB OTG in dual mode where I want to run USB device and Host on same application. So till now I couldn't find a single application or sample related to this, if there are any then please provide the reference here. If no then ,help me to narrow down the path for me how to achieve this. 

regards,
Shikamaru

ALABSTM, 2024-02-26:
Hi @Shikamru,

According to [this](https://community.st.com/t5/stm32-mcus-embedded-software/are-there-any-examples-of-stm32f769-dual-role-hs-usb-otg/m-p/176895#M10788) comment, it seems the STM32CubeF7 firmware does not provide the kind of applications your are looking for, indeed.

Your question has been forwarded to our development teams. I will get back to you as soon as I have their answer.

With regards,

ALABSTM, 2024-02-29:
Hi @Shikamru,

Back to you with more info. Our development teams confirm there is not such example showing the dual role of the USB OTG.

However, there is this demo, `STM32CubeH5/Projects/STM32H573I-DK/Demonstration/Demo` showing the dual role, but of the **USB-PD**. Below and extract of its "readme" file.

> This application demonstrates the functionalities of the Dual Role Power (DRP) and Dual Role Data (DRD) of the USB-PD protocol.

This demo has been removed from the `STM32CubeH5` repository on GitHub (you can refer to [this](https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer/discussions/17) post for the rationale). Luckily, it is available on [st.com](https://www.st.com/en/embedded-software/stm32cubeh5.html#get-software).

I hope this answers your question. Please allow me to close this thread.

With regards,

closed 2024-02-29 by ALABSTM