STMicroelectronics/STM32CubeF7 #117: STM32 USB OTG

Shikamru, 2024-03-05:
HI team,
I am using STM32F769 series controller(stm32f769 discovery board). I want to use USB_OTG_HS , as soon as I select this in STMCube IDE , middleware is getting disabled(which is fine). even I proceed further to generate the code , there is no low level driver also.  I checked all the samples provided from F7 firmware MCU package I couldn't see any sample. 
[image]

 As per the datasheet it is supported, however I couldn't find any example for the same.  one of the ticket it is clearly mentioned that F7 firmware doesn't have support for this.  please conclude.

Query -> [https://github.com/STMicroelectronics/STM32CubeF7/issues/116](url)

regards,
Shikamaru

ALABSTM, 2024-03-08:
Hi @Shikamru,

I confirm the behavior you reported. Your point has been forwarded to our development teams. I will keep you informed.

With regards,

ALABSTM, 2024-03-08:
ST Internal Reference: 175615

Shikamru, 2024-03-18:
HI @ALABSTM , Any update on this.

ALABSTM, 2024-03-18:
Hi @Shikamru,

A fix seems to have been implemented internally. I have contacted the development teams to ask for the version this fix will be released into. I will get back to you as soon as I have their answer.

With regards,

ALABSTM, 2024-03-22:
Hi @Shikamru,

The fix shall be part of release **1.12.0** of the CubeMX. We cannot share a date for the moment.

Since this issue does not affect any of the software published within this repository, please allow me to close this thread. Thank you for your comprehension and thank you again for your report.

With regards,

closed 2024-03-22 by ALABSTM