STMicroelectronics/STM32CubeF7 #118: core_cm7.h is out of date

kevinpt, 2024-04-11:
**Describe the bug**
The code in core_cm7.h is missing macros that are in present in the STM32CubeH7 version of the file. This STM32F7 repo has [V5.0.8 of the file from 2018](https://github.com/STMicroelectronics/STM32CubeF7/blob/294f7eb25253ec3fad0f0f80a4b13ac9a937a236/Drivers/CMSIS/Include/core_cm7.h#L1). The STM32H7 version is [V5.1.1 from 2019](https://github.com/STMicroelectronics/STM32CubeH7/blob/df18cb5dcd8cb20b83ed71c716f10a0abea94891/Drivers/CMSIS/Include/core_cm7.h#L2234).

In particular, the older copy is missing the macros __SCB_DCACHE_LINE_SIZE and __SCB_ICACHE_LINE_SIZE. There may be more updates that should be carried over as well.

The header in Drivers/CMSIS/Include/core_cm7.h and should be updated with the newer version from another repository such as STM32CubeH7.

PhilippHaefele, 2024-04-12:
I agree that the CMSIS (core) should be updated to a newer CMSIS version

TOUNSTM, 2024-04-16:
ST Internal Reference: 81352

TOUNSTM, 2024-04-16:
Hello,

The `core_cm7.h` file from `v5.1.1` is part of the CMSIS Core `v5.6.0`, which is integrated into the STM32H7 series. However, for the STM32F7 series, we're currently using CMSIS Core `v5.4.0_cm7` with the matching `core_cm7.h` file. Unfortunately, our development team has no plans to upgrade to the newer version of CMSIS at this time.

Please allow me to close this issue as "wontfix". Thank you for your understanding.

Best Regards,

closed 2024-04-16 by TOUNSTM