STMicroelectronics/STM32CubeF7 #120: Instructions for running example programs.

ibeckermayer, 2024-07-28:
I've purchased a `STM32F767ZI-Nucleo` and found my way to this repository in an attempt to get one of the example projects up and running. I see that there's a [directory of projects for my particular board](https://github.com/STMicroelectronics/STM32CubeF7/tree/master/Projects/STM32F767ZI-Nucleo).

However, what appears to be entirely missing are straightforward instructions on actually getting any of these up and running with any of the supported platforms. 

It seems like EWARM, MDK-ARM, and SW4STM32 are the most commonly supported, however oddly the STM32CubeIDE and STM32CubeMX makefile-style projects (which I'm familiar with) apparently aren't. This is odd given that those are ST products whereas the others I listed aren't. It also seems as though these example projects use the board's corresponding BSP, but don't themselves include it in their respective directories. I was able to find [this document](https://www.st.com/resource/en/user_manual/um2298-stm32cube-bsp-drivers-development-guidelines-stmicroelectronics.pdf) which details the layout of the BSPs, however it remains unclear precisely how to integrate that with any of the example projects.

The only official documentation I can find on getting an example project up and running is the following (borderline humorous in its non-specificity and un-helpfulness)

```
@par How to use it ?

In order to make the program work, you must do the following:
 - Open your preferred toolchain
 - Rebuild all files and load your image into target memory
 - Run the application
 ```

From my perspective, some basic documentation on how to get the project actually up and running on the physical board would be table stakes for a project like this. Surely the engineer who designed and built each project had to at least test it on the board itself using at least one of the supported toolchains -- could you not ask that person to document their project setup more precisely, so that your customers are able to straight-forwardly replicate their work and modify it for evaluation purposes?

ALABSTM, 2024-07-29:
Hi @ibeckermayer,

The  `SW4STM32` project files are easily convertible using the `STM32CubeIDE`. Just open the `.project` file of an example of your choice using `STM32CubeIDE` and follow the instructions in the popup messages. A few clicks and you should be done.

Actually, `SW4STM32` is the "ancestor" of `STM32CubeIDE`, as both are based on `Eclipse`.

With regards,

closed 2024-07-29 by ALABSTM

ibeckermayer, 2024-07-29:
@ALABSTM Thanks for the explanation. I did as described and the project appears to be building. Apologies for my pretentious tone, I was frustrated by my own ignorance.

ALABSTM, 2024-08-08:
Hi @ibeckermayer,

No worries. You're more than welcome.