STMicroelectronics/STM32CubeF7 #121: Incorrect Handling of TX Timestamp Flags in STM32 Ethernet Driver

marwaiehm-st, 2024-09-03:
The current implementation of the Ethernet driver appears to have been incorrectly copied from the H5/H7 HALs. Specifically, the flags ETH_DMATXDESC_LS and ETH_DMATXDESC_TTSS are being used incorrectly. These flags are from the DESC0 register, and the current implementation does not set the TX timestamp properly when required.

erwango, 2024-09-04:
See https://github.com/zephyrproject-rtos/hal_stm32/pull/232 for a fix proposal.

ASEHSTM, 2024-09-05:
See also https://github.com/STMicroelectronics/stm32f7xx_hal_driver/pull/9

ASEHSTM, 2024-09-05:
Fixed in [7737fc4](https://github.com/STMicroelectronics/stm32f7xx_hal_driver/commit/7737fc40e82e8e7574e4370979d2f7f5e8b8ff28)

closed 2024-09-05 by ASEHSTM