STMicroelectronics/STM32CubeF7 #123: UART DMA in IDLE Mode calls callback function twice in Half transfer

Memarnejad, 2025-02-11:
Hi,
**Describe the set-up**
Develop a program in which UART Receive is in RX **IDLE** Mode Detection and uses **DMA** in **Circular** Mode for data transfer

**Describe the bug**
In the **middle** of transfer, HAL_UARTEx_RxEventCallback is called **twice**. once for _DMA half complete transfer_, another one for _IDLE_ detection. Although it's logical to detect both events, But I guess this It's not the intention of HAL developers, since callback is not called twice at the end of transfer (one for complete, another one for IDLE). So It's a bug base on this intention.

**How To Reproduce**
1. Put UART reception in RX IDLE mode detection using HAL_UARTEx_ReceiveToIdle_DMA()

2. Set the UART DMA in Circular Mode

3. Start Sending data to UART with gap in between to trigger IDLE Mode

4. if the transfer size is Even, the twice call is occured in exactly middle of transfer ((e.g.  transferSize = 8; Problem occurs in 4th Byte, both half and idle events with 4 as Size parameter to HAL_UARTEx_RxEventCallback)

5. if the transfer size is ODD, the twice call is occured in ceiling of middle of transfer ((e.g.  transferSize = 7; Problem occurs in 4th Byte, Half callback with 3 and Idle callback with 4 as Size parameter to HAL_UARTEx_RxEventCallback)

**How to Fix**
In stm32f7xx_hal_uart.c file, in HAL_UART_IRQHandler function, around line 2335, in IDLE line interrupt handling add another condition for circular mode:
```
 ((huart->hdmarx->Init.Mode != DMA_CIRCULAR) || (nb_remaining_rx_data != huart->RxXferSize / 2U))
```
to the below code:
```
if ((huart->ReceptionType == HAL_UART_RECEPTION_TOIDLE)
      && ((isrflags & USART_ISR_IDLE) != 0U)
      && ((cr1its & USART_ISR_IDLE) != 0U))
  {
    //Some Other Stuff....
    //Some Other Stuff....
      uint16_t nb_remaining_rx_data = (uint16_t) __HAL_DMA_GET_COUNTER(huart->hdmarx);
      if ((nb_remaining_rx_data > 0U)
          && (nb_remaining_rx_data < huart->RxXferSize)
		  && ((huart->hdmarx->Init.Mode != DMA_CIRCULAR) || (nb_remaining_rx_data != huart->RxXferSize / 2U)))
      {
       //Some Other Stuff....
       //Some Other Stuff....
      }
       //Some Other Stuff....
       //Some Other Stuff....
  }
```
The above Patch fixes the call of both IDLE event and Half Complete for transfer.

Another problem is that when transfer size is ODD, half complete callback calls HAL_UARTEx_RxEventCallback with (somehow) wrong Size parameter (e.g. transferSize = 7; call back is called with floor(7/2) = 3, but this size was informed to callback in previous call to HAL_UARTEx_RxEventCallback with IDLE Event and ceil(7/2) = 4 MUST be informed now). 
To Fix this issue, in UART_DMARxHalfCplt function, use the ceil value:
```
  if (huart->ReceptionType == HAL_UART_RECEPTION_TOIDLE)
  {
    HAL_UARTEx_RxEventCallback(huart, ceil(huart->RxXferSize / 2.0));
  }
```
With above two patches, now for even transfer size, only half complete is called for transferSize / 2, and for odd transfer size, only half complete is called for ceil(transferSize / 2)

Hope this fix helps others as well
best regards

KRASTM, 2025-02-14:
Hello @Memarnejad,

Thank you for the report.

After a first analysis, I do not quite agree with you on one point: among some changes to the driver, we have one that aims to
detect the end of transmission (i.e. get the IDLE notification) even if received length is same as reception buffer size.

For example: using HAL_UARTEx_ReceiveToIdle_DMA() with Circular mode and in case of reception of 20 bytes (size of data to be received) then pause, you will have:
    - HT (10 bytes): Callback will retrieve 10 data from reception buffer (pos 0-9) and rx_index will be set to 10
    - TC (20 Bytes): Callback will retrieve 10 data from reception buffer (pos 10-19) and rx_index will be set to 20
    - IDLE (20 bytes): Callback will not retrieve any new data from reception buffer as rx_index is already set to 20

Just to be sure, for steps 4 and 5 you have sent 4 bytes with a pause? and after you have transmitted the remaining data?

So, I think it's normal `if the transfer size is Even` to have HT and Idle. For the odd size case, I will run some tests and check with our team, in the meantime, could you please share your config and more details for further analysis.

With regards,

Memarnejad, 2025-02-14:
Hi,
For my test, I send every bytes with a pause after to see if events(Half, Full and Idle) occurs correctly or not?

Consider transfer size of 8 (even):
* For bytes 1 to 7 except 4 only Idle
* For byte 4 both half and idle
* For byte 8 only full

Consider transfer size of 7 (odd):
* For bytes 1 to 6 except 4 only Idle
* For byte 4 both half and idle
* For byte 7 only full

What is not consistent is that if the intention was to send the user the idle event, irrespective of half and full, then for both bytes (4 and 8 even ... 4 and 7 odd), we Must have had that event. But only for half event this idle event occurs. So I guess this a bug as base on this intention we must have had idle with full event as well.

In my opinion, idle event is a way to inform user whether there are data or not, so if the half or full event occur, the user will get informed of status, so we may remove idle event in case there is half or full at the same time with idle.
Of course, we can propagate all events to HAL users and let them handle them (just as you say about circular buffer pointer) , but the goal of using HAL is to remove some details and handle them on behalf of users

Regards

KRASTM, 2025-02-18:
Hello @Memarnejad,

I ran a test with an example on **STM32F767ZI Nucleo board**, and it works as expected, below a screenshot for the 2 cases:

- Transfer size of 8: I send a byte with a pause, an `A` and a `P` and I recover each event via HAL_UARTEx_GetRxEventType(), so for Idle event "I", for Half transfert "H" and for transfert Complet "T".

[image]

As you can see from the image, we have
       - For byte 4 both Half and Idle
       - For byte 8 both TC and Idle

- Transfer size of 7: same scenario and the same result:

[image]

I suggest you check and verify that you have the latest **HAL driver**, your config especially UART DMA (if it's possible share it with us).

There is an example on WB Serie, you can check it for more details, under this path `P-NUCLEO WB55.Nucleo\Examples\UART\UART_ReceptionToIdle_CircularDMA`

With regards,

closed 2025-02-24 by KRASTM